brainstorm-tools/brainstorm3 #738: Using fieldtrip functions for 3DScanner

chinmaychinara91, 2024-09-25:
For importing an OBJ into brainstorm:
1. The `in_tess_wftobj` function in actually a slightly modified submodule of https://www.fieldtriptoolbox.org/reference/fileio/ft_read_headshape. Should we use the fieldtrip function directly or since the license allows us, just continue to do it as it is right now (and just put a reference to the fieldtrip function).
1. We also use the https://www.fieldtriptoolbox.org/reference/forward/ft_convert_units function to determine the units of the vertices based on the shape of the head. This is a good function to have.

Both these functions are available as part of spm12 plugin (/external/fieldtrip) and also the fieldtrip plugin. How do we go about this ?

rcassani, 2024-09-25:
Are not these topics part of the open PR #716?

If so, please close this issue, and add the question in the PR conversation to have all the information in one place instead of having spread across PRs and Issues.

chinmaychinara91, 2024-09-25:
ok doing that.

chinmaychinara91, 2024-09-25:
posting in PR https://github.com/brainstorm-tools/brainstorm3/pull/716 to keep it consolidated in one thread.

closed 2024-09-25 by chinmaychinara91