brainstorm-tools/brainstorm3 #741: Inconsisten behaviour on macOS (Intel) of `direct_pac_mex.c`

rcassani, 2024-10-04:
**Description**
The [provided .mex files](https://github.com/brainstorm-tools/brainstorm3/tree/master/toolbox/connectivity/private) to compute PAC do not behave the same on macOS using Intel CPU.

**To reproduce**
Run the script [`tutorial_resting`](https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/script/tutorial_resting.m) on a macOS (Intel) system, using the options `ismex = 1` and `parallel = 0` in the process `process_pac`.

**Expected behaviour**
On line 181 of [`bst_pac`](https://github.com/brainstorm-tools/brainstorm3/blob/7ef5e6fef78bc41bec942506065abb942e919532/toolbox/connectivity/bst_pac.m#L181):
```
DirectPAC = direct_pac_mex(permute(PHASE, [2,3,1]), permute(AMP, [2,3,1]));
```

the function `direct_pac_mex` should return a complex double matrix of shape `[16,19]` (it does in Linux and Windows). 
However in macOS (Intel) it returns a complex double matrix of shape `[0, 2177491158568009729]`.

rcassani, 2024-10-04:
For the moment this is avoid in the the [`tutorial_resting`](https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/script/tutorial_resting.m) by setting `ismex` to `0` in e2ef5581e64be6c647e76e291dbfb28f3ef188ef

Edouard2laire, 2024-10-07:
hi. 

If you want to export the PHASE and AMP matrix, I can recompile the mex on my side and do some test. (also recompiling for ARM mac)

rcassani, 2024-10-07:
Thanks for the help @Edouard2laire,

You can find the test data in this link:
https://www.dropbox.com/scl/fi/ghklrig4udz3lfnn81htf/bst_pac_test_data.mat?rlkey=iv2net9o8lks3d98scabttk58&dl=0

It contains `PHASE` and `AMP` and other potentially useful variables. Also, the variable `DirectPACGoodResult` contains the expected result with the following command using the `mex` file
```
DirectPAC = direct_pac_mex(permute(PHASE, [2,3,1]), permute(AMP, [2,3,1]));
```

It can be compared against the result obtained with the `non-mex` implementation as well:
```
for ihf = 1:length(hfreq)
    DirectPAC(:,:,ihf) = reshape(sum( bst_bsxfun(@times, PHASE, AMP(:,:,ihf)), 2 ), [nSignals, length(lfreq)]);
end
% Permute to match the result from the mex 
DirectPAC = permute(DirectPAC, [2,3,1])
```

commit 7199f3a3 (2024-10-15): Recompile mex file for PAC (fix #741) (#744)

Fixes #741

* Add mex for ARM
* Add Intel version
* Script: PAC, use Matlab`.mex` implementation

---------
closed 2024-10-15 by rcassani