brainstorm-tools/brainstorm3 #782: [Feature Request] Support for importing Neuroscan MEG (.dat) files

jade-l228, 2025-04-15:
Dear Brainstorm Team,

I am a university professor, and our institution has recently purchased a MEG system from **Neuroscan**. We are currently in the process of establishing a data analysis workflow and would like to use **Brainstorm** as the main analysis software.

However, the MEG data from the Neuroscan system is saved in `.dat` format, and it seems that Brainstorm does not currently support this type of file. We have tried importing it through various options in the GUI, but none succeeded.

I would like to ask:
- Is there any current way to import `.dat` files from Neuroscan MEG systems into Brainstorm?
- If not, would the development team consider adding this compatibility in future versions?
- We are happy to provide sample data files to help with testing and development, if needed.

We would greatly appreciate your support, as Brainstorm would be a valuable tool for both our research and teaching purposes.

Thank you for your time!

Best regards, 
Jade

rcassani, 2025-04-17:
Hello @jade-l228, congratulations for the new MEG!

- You are right, currently the Neuroscan MEG data is not supported.
- Yes, it is the future developments to support this Neuroscan MEG
- Please provide a couple of sample data files

jade-l228, 2025-04-30:
Thank you very much for your reply. I apologize for the delay in getting back to you. Attached is the MEG data for the P1 component that we collected, 

https://www.dropbox.com/scl/fo/ycqk5djvgcltt6brwqq67/AKwYdv4rkdfYqt3C38_Xl74?rlkey=dalf02egxius1f4xhsbxw5i4m&amp;dl=0

and we hope it will be helpful. We are all looking forward to using Brainstorm to analyze this type of data. Thank you again.  

Regards,
Jade

发自我的iPhone

jade-l228, 2025-07-30:
Hi brainstorm support team,
Here is a new link: 
https://www.dropbox.com/scl/fo/ycqk5djvgcltt6brwqq67/AKwYdv4rkdfYqt3C38_Xl74?rlkey=dalf02egxius1f4xhsbxw5i4m&amp;dl=0

Thanks,
Jade

 
@jade-l228, the provided link is broken

rcassani, 2025-08-01:
@jade-l228, the second and first links are the same and they do not work

jade-l228, 2025-08-02:
Sorry, may I ask what kind of transmission method we should use for you to open it? We have checked that this Dropbox link is accessible, 
and we also use it for transmission and communication with foreign parties.

Thanks,
Jade

jade-l228, 2025-08-04:
> @jade-l228, this is what I get from your link:
>  image.png (view on web)
> Can you upload it in other place?
>  If not, please write me an email, and I can provide you with a link where to upload the data

Sorry, maybe other links of China don’t work, please give me a a link I can upload the data. Thanks for your patience. 

Best regards 
Jade

发自我的iPhone

rcassani, 2025-08-04:
Sure, I can provide you the link.
Please send me an email to `raymundo.cassani[at]mcgill.ca`

jade-l228, 2025-08-05:
I have sent you an email ,Look foward to your reply !

Thanks,
Jade

         Original

Sent Time:2025-08-4- 22:34