brainstorm-tools/brainstorm3 #793: EEG points from Polhemus file not converted from CTF coordinates to SCS

Moo-Marc, 2025-05-12:
Bug found by Ellie in our lab.

First, when loading a CTF dataset with .pos file that includes EEG coordinates, they are not saved in Bst.  I changed this part of the code recently, but this was already the case as it only keeps the .Headpoints, while the EEG positions are saved in .Channels:
https://github.com/brainstorm-tools/brainstorm3/blame/b85498b157a4db15d01e66966ad14872b40eb43f/toolbox/io/in_fopen_ctf.m#L190

Second, when adding EEG positions, it looks like import_channel no longer applies the CTF to SCS transformation, though it should do it here, unless the check is incorrect in this case:
https://github.com/brainstorm-tools/brainstorm3/blob/b85498b157a4db15d01e66966ad14872b40eb43f/toolbox/io/import_channel.m#L549

A bit more testing and debugging needed.

rcassani, 2025-05-12:
Hi @Moo-Marc, can you share an simple CTF dataset with MEG and EEG to test and address those two points?

rcassani, 2025-05-21:
PR #797 aims to solve it. 
@Moo-Marc, can you take a look?