brainstorm-tools/brainstorm3 #802: Missing EEG Channel Names When Importing FIF/BIDS Files from NeuroMag

tmedani, 2025-05-28:
Problem Summary:

Multiple centers have encountered a recurring issue when importing .fif files into Brainstorm: the EEG channel names are not properly retained or imported. This is particularly evident with data originating from NeuroMag systems, where the channel labels either appear as generic names or are entirely missing.

Suggested Solutions:

To address this issue and improve interoperability with .fif data:

1. Enhancement During Import:

Add an option during the import process to replace standard channel names using a user-specified mapping file (e.g., .csv or .txt).

This file would contain a simple mapping (e.g., EEG001 → Fp1, EEG002 → Fz, etc.).

2. Post-Import Channel Editing:

Add a right-click context menu option on the channel file (in the Brainstorm GUI) to "Replace Channel Names from Mapping File".

This would allow retrospective correction of channel names after the data has already been imported.

tmedani, 2025-06-11:
Thanks @yashvakilna 

@chinmaychinara91 can you please check this and compare it with your current scripts?

chinmaychinara91, 2025-06-17:
Based on the discussion with with @yashvakilna and Michelle today and adding to the yash's https://github.com/brainstorm-tools/brainstorm3/issues/802#issuecomment-2963851225 above:

When importing FIF files through `process_import_bids`, for non-MEG (here EEG) channels, two post processing steps are required:

1. Renaming the `EEGxxx` channel names to the proper ones as provided in the `_channels.tsv` file
2. Update the locations for the EEG channels from `_electrodes.tsv`. The reason we do this is because the locations in the FIF file may have error but `_electrodes.tsv` will have the most accurate locations.

rcassani, 2025-08-01:
@yashvakilna, @chinmaychinara91, @tmedani, was this was solved with #810?
If so we can close the issue.

closed 2025-08-01 by rcassani